CitizenSensor/CitizenSensor #43: Bestückung der Platinen (Raspberry Pi HAT/ Electrometer)

Hswts, 2020-04-22:
Wir  haben anhand der Bilder eine Bestellliste für den Raspberry Pi HAT erarbeitet. (siehe Anhang) Wir sind uns aber nicht ganz sicher ob die so vollständig ist. 
[Stückliste HAT.docx](https://github.com/CitizenSensor/CitizenSensor/files/4518487/Stuckliste.HAT.docx)

Zur Bestückung des Electrometer Board fehlen uns leider jegliche Infos könnt ihr da was hochladen ?

Uns ist bei manchen Bauteilen ein n.b. aufgefallen was bedeutet dieses?

drayde, 2020-04-23:
@matthias6 könntest Du mal die Stückliste anschauen?

Wo steht denn das "n.b.", @Hswts ? Hab es jetzt nicht gesehen.

Electrometer Board: wir sind dran

matthias6, 2020-04-23:
n.b beudeutet nicht bestücken

Hswts, 2020-04-27:
@matthias6  habt ich noch eine Liste mit der Bestückung von dem Elektrometer-Board ?

matthias6, 2020-04-27:
[image]
Wir werden die fehlende Doku für das Elektrometer-Board hoffentlich bald hochladen. Ich hab hier noch von einem veralteten Schaltplan die Schaltungen, die sich glaub nicht geändert haben. R3-R6 sind 100 kOhm, C2 und C3 1 µF jeweils 0805.
Ich schau morgen, ob ich auch einen Bestückungsplan finde.

matthias6, 2020-04-28:
[image]
gefunden

Hswts, 2020-04-29:
Was für ein Bauteil ist das SJ1 ?

matthias6, 2020-04-30:
JP1 ist eine headerleiste einreihig, 3 Kontakte, rastermaß 2.54 mm
SJ1 ist kein Bauteil. SJ = Solder Jumper. Hier müssen zwei der drei Pads mit Lötzinn verbunden werden.

matthias6, 2020-04-30:
PCI E 36 slot ist der Footprint für den PCIeX1-Stecker, das ist kein Bauteil.
TPB2,54 hört dich an als wäre es ein Testpoint, um Spannung mit dem Multimeter abzugreifen beim Testen (habs nicht gefunden, wo steht das?)
MCP48X1SN das kann ein MCP4801 sein (8-bit DAC sollte hier reichen).

matthias6, 2020-04-30:
Hier ist alles gelb markiert, welche Bauteile man unbedingt für den CitizenSensor braucht. Die restlichen sind für andere Anwendungszwecke: 
[image]

matthias6, 2020-04-30:
Achtung: Der AD8226 darf nicht bestückt sein, wenn die Platine im CitizenSensor zum Einsatz kommen soll.